ros2/ros2_documentation #5214: 📝Copy To clipboard button not working

obamwonyi, 2025-04-05:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Description

In the page [link](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#setup-colcon-tab-completion) the copy to click board button is not working, not sure why but when I clicked it, it gave the prompt copied but when I pasted nothing was pasted. the image below shows the exact section that is not copying

[image]

### Suggested Fix

I could fix it if I knew where the implementation was done. I am familiar with Sphinx.

christophebedard, 2025-04-05:
Thanks for reporting this! I can reproduce the issue on the website and also when building the docs locally.

Clicking on the copy button for the code blocks above and below correctly copies the commands, but it indeed doesn't work for the middle code block. All 3 [code blocks](https://www.sphinx-doc.org/en/master/usage/restructuredtext/directives.html#directive-code-block) use the `console` language, which expects _command_ lines to start with a `$` and _output_ lines to not start with a `$`. Clicking on the copy button should then only copy the command lines. This is a change I recently made, since the copy button would previously copy everything in a `console` code block (command lines with the `$` and output lines), see #5112. Therefore, all of these lines should have a `$` if they use `console`. For more information, see [this section of the "Contributing to ROS 2 Documentation" article](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html#code-blocks-bash-vs-console).

However, that doesn't explain why the code blocks above and below can be copied. They have only a single line, while the code block for which the copy button doesn't work has 2 lines: https://github.com/ros2/ros2_documentation/blob/65ac7e81baffd8499bce8f94fb4a1fcd2653f0ac/source/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.rst#L360-L377

Therefore, I assume that the copy button just assumes that a single line (but not two lines) in a `console` code block is a command line even if it doesn't have a `$`. Overall, I believe all 3 code blocks here are "wrong" and should either (1) be changed to `bash` or (2) keep `console` but add a `$` to the beginning of each line. Both of these solutions lead to the copy button correctly copying the commands when I build the docs locally.

I think many other code blocks have this issue. For example, the code block under the ["Setup `colcon_cd`" section](file:///home/christophe.bedard/ros2_documentation/build/html/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#setup-colcon-cd) also has the same issue. I'll clarify [the instructions](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html#code-blocks-bash-vs-console), because this issue proves that this sentence about `console` code blocks isn't true (i.e., leads to the copy button not working):

> In general, the prompt symbol (`$`) can be omitted if the code block does not contain any output lines.

For code blocks with just commands without any output, we could just use `bash`. That's probably the simplest fix we can make right now. We should go through the docs and change `console` code blocks to `bash` if they only have command lines (not output lines) _and_ don't use `$`. Your help would be appreciated! I would recommend starting with the `rolling` branch. You can find documentation on how to build the docs locally [in this section](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html#building-the-site-locally).

christophebedard, 2025-04-05:
As for this:

> Then there is also another error in this page

Can you clarify why you think this is an error?