ros2/ros2_documentation #5226: 📝 <Brief Title>

fujitatomoya, 2025-04-12:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Rolling

### Description

following `rosdep install` command during source build environment setup described in https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html, failed

```console
root@tomoyafujita:~/ros2_ws/colcon_ws# rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
rosidl_generator_tests: Cannot locate rosdep definition for [python3-fastjsonschema]
rti_connext_dds_cmake_module: Cannot locate rosdep definition for [rti-connext-dds-7.3.0]
```

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html

### Additional Context

https://github.com/ros2/ci/pull/811

nuclearsandwich, 2025-04-12:
```
rosidl_generator_tests: Cannot locate rosdep definition for [python3-fastjsonschema]
rti_connext_dds_cmake_module: Cannot locate rosdep definition for [rti-connext-dds-7.3.0]
```

Both of these are currently defined on master:
- https://github.com/ros/rosdistro/blob/d1173c022c4c79e685c6f4266995b59f596f1154/rosdep/python.yaml#L5918
- https://github.com/ros/rosdistro/blob/d1173c022c4c79e685c6f4266995b59f596f1154/rosdep/base.yaml#L8477

So it may be that your rosdep sources are out-of date?

fujitatomoya, 2025-04-12:
i confirmed that the error can be reproducible.
something is wrong with `rti-connext-dds-7.3.0-ros` debian package.

```console
root@2f0563ab2b53:~/ros2_ws/work# rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"
...<snip>

The installation will be aborted if you continue without accepting.

After a successfull installation, a copy of the license agreement will be saved at
/opt/rti.com/rti_connext_dds-7.3.0/RTI_License_Agreement.txt

Do you accept the terms of the RTI License Agreement? [yes/no] yes

config: RTI license agreement for non-commercial use accepted for all users of this system.
/var/lib/dpkg/tmp.ci/config: 280: Syntax error: "fi" unexpected
dpkg: error processing archive /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb (--unpack):
 new rti-connext-dds-7.3.0-ros package pre-installation script subprocess returned error exit status 2
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 79.)
debconf: falling back to frontend: Readline
Errors were encountered while processing:
 /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y rti-connext-dds-7.3.0-ros] failed
```

the same error happens with `no`.

```console
...<snip>
Do you accept the terms of the RTI License Agreement? [yes/no] no

config: RTI license agreement for non-commercial use NOT accepted, stopping installation...
/var/lib/dpkg/tmp.ci/config: 280: Syntax error: "fi" unexpected
dpkg: error processing archive /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb (--unpack):
 new rti-connext-dds-7.3.0-ros package pre-installation script subprocess returned error exit status 2
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 79.)
debconf: falling back to frontend: Readline
Errors were encountered while processing:
 /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y rti-connext-dds-7.3.0-ros] failed
```

fujitatomoya, 2025-04-12:
@lobolanja @fgallegosalido this is breaking the development environment setup, any thoughts?

christophebedard, 2025-04-12:
As far as these installation instructions go, we explicitly want to skip installing Connext, since it's not the default and should be installed later if the user wants it. Therefore, we should update the command to ignore `rti-connext-dds-7.3.0` instead of `rti-connext-dds-6.0.1`. So, for now, we can make this change to fix this specific tutorial, and then we can look into why it fails to be installed.

Also, as far as I can tell, we've ignored the rosdep key for Connext ever since the ROS 2 installation/build instructions were moved from the original wiki to the docs (2018): https://github.com/ros2/ros2_documentation/blob/99bbb6291b339a8fd6293ed832dc68604aa46d44/Linux-Development-Setup.md#install-dependencies-using-rosdep. And I believe we've only been manually installing it, e.g., https://docs.ros.org/en/rolling/Installation/DDS-Implementations/Working-with-RTI-Connext-DDS.html#install-rti-connext-dds.

I can reproduce the issue in a fresh `ubuntu:noble` container:

```
Do you accept the terms of the RTI License Agreement? [yes/no] yes

config: RTI license agreement for non-commercial use accepted for all users of this system.
/var/lib/dpkg/tmp.ci/config: 280: Syntax error: "fi" unexpected
dpkg: error processing archive /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb (--unpack):
 new rti-connext-dds-7.3.0-ros package pre-installation script subprocess returned error exit status 2
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 79.)
debconf: falling back to frontend: Readline
Errors were encountered while processing:
 /var/cache/apt/archives/rti-connext-dds-7.3.0-ros_7.3.0.0-5-3bullseye_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y rti-connext-dds-7.3.0-ros] failed
```

Installing version 6.0.1 through rosdep does work, but it requires user interaction if you follow the instructions as-is, which is a bit unusual.

christophebedard, 2025-04-12:
> Therefore, we should update the command to ignore `rti-connext-dds-7.3.0` instead of `rti-connext-dds-6.0.1`. So, for now, we can make this change to fix this specific tutorial,

See #5227

lobolanja, 2025-04-12:
Hi @fujitatomoya 

I’ll take a look at this with the team on Monday morning (Spain time). In the meantime, it should be fine to skip rti-connext-dds-7.3.0 as @christophebedard suggested.

Thanks for flagging the issue!

lobolanja, 2025-04-14:
Hi @fujitatomoya @mjcarroll @christophebedard and @nuclearsandwich ,

We’ve reviewed the package and identified a typo in the script responsible for handling the license acceptance prompt during installation. We’re currently rebuilding the package with the fix applied.

In the meantime, setting the environment variable to accept the license will serve as a workaround.

We’ll let you know as soon as the updated package is ready, so the corresponding bits in the ROS 2 upstream repositories for the RTI Connext Debian package can be updated.

Thanks again for your support!

lobolanja, 2025-04-18:
Hi @fujitatomoya , @mjcarroll , @christophebedard and @nuclearsandwich ,

We've already provided the updated package with the fix applied for the license prompt handling.

That said, we understand that the current behavior—still requiring user interaction to accept the license—might not fit well with the release process at this stage. If skipping the 7.3.0 package for now is the best path forward, we're aligned with that.

Looking ahead, now that we have a dedicated team at RTI actively working on ROS and RMW improvements, we're committed to streamlining this experience for future releases. We'd be happy to collaborate more closely to make the integration smoother and more predictable moving forward.

Thanks again for all the support and coordination!

Best,
Juanca

christophebedard, 2025-04-29:
Yeah, the rosdep key is now correctly skipped, so this issue can be closed.

closed 2025-04-29 by christophebedard