ros2/ros2_documentation #5229: 📝 Conflicting "option Signed-By" caused by Ubuntu Deb Packages Install Procedure

ChrisE2018, 2025-04-12:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Jazzy

### Description

In the documentation page <https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html#id5> there is a step labelled "Then add the repository to your sources list." with an echo command to write a file. 

I think this step is incorrect and should be removed.

The symptom is the build problem described here: <https://answers.ros.org/question/410123/>

When I skip this step (inside a docker container) I get a working ros2 environment. When I follow the procedure as shown my build fails with the error message:

"Conflicting values set for option Signed-By regarding source http://packages.ros.org/ros2/ubuntu/ noble: /usr/share/keyrings/ros2-latest-archive-keyring.gpg != /usr/share/keyrings/ros-archive-keyring.gpg
0.155 E: The list of sources could not be read"

ChrisE2018, 2025-04-12:
For my install, I am running inside a docker container using instructions from <docker pull osrf/ros:jazzy-desktop>

fujitatomoya, 2025-04-12:
please try with ubuntu noble as described in https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html#resources, i am not seeing any error.

closed 2025-04-12 by ChrisE2018