ros2/ros2_documentation #5244: Bumb CI to 24.04

Nils-ChristianIseke, 2025-04-19:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Migration from Ubuntu 22.04 to 24.04

### Description

We currently pin dependencies to Ubuntu 22.04 system packages (like python3-stevedore 3.5.0). Upgrading to 24.04 requires updating all dependencies and CI infrastructure.

Nils-ChristianIseke, 2025-04-19:
I can do that, but I won't start working on this before may. So if anyone wants to do it now, go ahead

fujitatomoya, 2025-04-19:
@Nils-ChristianIseke thanks for creating issue.

I do not we are in rush on this, so i will go ahead to assign this to you as offered.

christophebedard, 2025-04-19:
Yeah, we're in no rush!

Before actually changing anything, I'd recommend figuring out what needs to be changed (versions of dependencies, Dockerfile, GitHub Actions, documentation about how to build the docs, etc.) and then figuring out what the concrete changes should be, especially the versions (based on the system package on Ubuntu 24.04).

Like I mentioned in https://github.com/ros2/ros2_documentation/pull/5241#issuecomment-2816017640, we'll also have to coordinate to update the nightly CI job that builds the docs, since that's separate from this repo. That might take some time since this isn't really a priority. I will ask about this at the next Infrastructure PMC meeting.

christophebedard, 2025-04-29:
FYI @Nils-ChristianIseke I can now trigger test builds for the docs (https://build.test.ros2.org/job/doc_ros2-documentation/), so I'm ready once you've updated the Dockerfile

Nils-ChristianIseke, 2025-05-09:
Working on this in #5595