ros2/ros2_documentation #5384: Windows Binary install instruction doesn't run

knmcguire, 2025-04-27:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Rolling

### Description

Hi!

If you install the windows according to the binary install instructions (as in [here](https://docs.ros.org/en/rolling/Installation/Windows-Install-Binary.html)), running the examples will fail because the prerelease binary zips are build in an environment where the workspace directory is called C:/pixi_ws, so anything that uses a python script to run from the terminal (and that's 'ros2' itself already) will fail due to that (I just get a `failed to create process`. 

After some discussions on discord together with @traversaro we found it's due to the shebang lines on top of the python scripts being hardcoded on a pixi_ws workspace.

### Suggested Fix

@traversaro had some notes on how tofix it:

1. "Add a post-install script to the windows binary archive that just looks for all the Scripts/*-script.py files and fixes the shebang line to be the correct one. This adds a step for the user, but it is probably the easiest option"
2. "Add a python.exe trampoline executable in the parent folder of the Scripts folder, that just launches the actual python.exe . This does not adds a step for the user, but it is not so easy and it is probably error prone"
3. "Go back to only support a specific hardcoded install prefix for the pixi env, not a big fan of it but definitely the solution that requires less work"

I think that 1 will be the best approach here as well

traversaro, 2025-04-27:
Related issue in ros2 repo: https://github.com/ros2/ros2/issues/1675 .

knmcguire, 2025-08-12:
Also for reference, there is a PR out that should make it much easier to install. https://github.com/ros2/ci/pull/817

knmcguire, 2025-10-16:
Ah yes... I would imagine... I have pushed for the other [PR ](https://github.com/ros2/ci/pull/817) to go through last week but I imagine ROSCon is delaying it now. If that goes through, than the full binary install should be simpler and folder agnostic.