ros2/ros2_documentation #5397: 📝 Windows installation from source instructions out of date

mikelikesrobots, 2025-04-28:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Jazzy

### Description

Latest instructions for building ROS 2 from source for Windows uses pixi as a package manager, but this is missing from the Jazzy documentation. Is it possible to backport the instructions?

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Installation/Alternatives/Windows-Development-Setup.html, https://docs.ros.org/en/jazzy/Installation/Alternatives/Windows-Development-Setup.html

christophebedard, 2025-05-03:
I believe that we do want to backport the Windows/Pixi instructions to the Jazzy docs, but haven't had the time to do so yet. Help is always welcome!

christophebedard, 2025-05-06:
Great! Let us know if you have any questions. I believe the main PR that needs to be backported to the `jazzy` branch is #4989. Then it would be good to compare the result (after backporting that PR) to the current Rolling docs to see if anything else is missing.