ros2/ros2_documentation #5416: 📝 Connext RMW binaries for Rolling use Connext 6.0.1 instead of 7.3.0

fgallegosalido, 2025-04-29:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Rolling

### Description

The Deb package `ros-rolling-rmw-connextdds` downloads Connext 6.0.1 instead of 7.3.0, which is what is expected in Rolling (and unreleased Kilted). It also downloads a Connext RMW binary compiled with 6.0.1 instead of 7.3.0.

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Installation/RMW-Implementations/DDS-Implementations/Working-with-RTI-Connext-DDS.html

### Screenshots or Examples (if applicable)

Command output for installing Connext RMW in Rolling:
```
$ sudo apt update && sudo apt install -q -y ros-rolling-rmw-connextdds
Hit:1 http://archive.ubuntu.com/ubuntu noble InRelease                                                    
Hit:2 http://security.ubuntu.com/ubuntu noble-security InRelease                                          
Hit:3 http://archive.ubuntu.com/ubuntu noble-updates InRelease                                            
Hit:4 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Hit:5 http://packages.ros.org/ros2/ubuntu noble InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
7 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  ros-rolling-rmw-connextdds-common ros-rolling-rti-connext-dds-cmake-module rti-connext-dds-6.0.1
The following NEW packages will be installed:
  ros-rolling-rmw-connextdds ros-rolling-rmw-connextdds-common ros-rolling-rti-connext-dds-cmake-module rti-connext-dds-6.0.1
0 upgraded, 4 newly installed, 0 to remove and 7 not upgraded.
Need to get 154 MB of archives.
After this operation, 552 MB of additional disk space will be used.
```

### Suggested Fix

Modify the deb package `ros-rolling-rmw-connextdds` to include a Connext RMW compiled with Connext 7.3.0, along with Connext 7.3.0 instead of 6.0.1.

### Additional Context

This issue was found while following the documentation, so I'm opening the issue here, but the fix must be done in other ROS 2 repository responsible of Deb packages.

fgallegosalido, 2025-04-29:
I'm not entirely sure how the deb packages are generated. Is there any repo/info I can check to see the process?

christophebedard, 2025-04-29:
You can see here that Rolling currently has version 1.0.0 but that version 1.2.0 (which includes the switch to Connext 7.3.0) is in the testing repo: https://repo.ros2.org/status_page/ros_rolling_default.html?q=SYNC.

So like Tomoya said, we just need a Rolling sync.

fujitatomoya, 2025-04-30:
@fgallegosalido So am I, there are still many things behind the scene for me too 😄 that is fun and exciting.

fgallegosalido, 2025-04-30:
@fujitatomoya right, I'll test the binaries whenever they are ready, and close this task right after the testing is done

fgallegosalido, 2025-05-05:
This can be closed as the binaries now download the correct Connext version along with the rmw compiled using 7.3.0.