ros2/ros2_documentation #5457: Document subordinate nodes aka sub-nodes

christophebedard, 2025-05-01:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [x] 🚀 Enhancement

### Distribution

all

### Description

I realized that subordinate nodes are not mentioned anywhere in the docs here. See: https://github.com/ros2/ros2_documentation/pull/5426.

### Suggested Fix

Consider documenting sub-nodes:

1. Explain the concept of sub-nodes in the node concepts page: https://docs.ros.org/en/rolling/Concepts/Basic/About-Nodes.html
    - This might be too advanced for this basic concepts page, though
2. Explain **why/in what situations sub-nodes might be useful** and show how they're used in a tutorial or how-to guide
    - I think it might make more sense as a tutorial
    - Maybe a "Using sub-nodes" intermediate tutorial, similar to the "Using the Node Interfaces Template Class (C++)" tutorial: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Using-Node-Interfaces-Template-Class.html

### Additional Context

The best documentation we have right now for sub-nodes is the `rclcpp::Node::create_sub_node()` documentation: https://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1Node.html#classrclcpp_1_1Node_1a8d5b012d7366795fbc8bb315be13a51b

See the original PR: https://github.com/ros2/rclcpp/pull/581. It's interesting to note that it comes from wanting to add a "NodeHandle"-like concept like ROS 1 had: https://github.com/ros2/rclcpp/issues/531.

SuperGops7, 2025-05-05:
Hey @christophebedard I can help with this issue

fujitatomoya, 2025-05-05:
@SuperGops7 all yours, thanks!

closed 2026-10-30 by ahcorde