ros2/ros2_documentation #5462: 📝 <Brief Title> Instructions on installing Debian Binary package may not work in zsh shell

Juliaj, 2025-05-02:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

kilted/rolling

### Description

Some commands to install Debian binary on Ubuntu may fail in `zsh` shell. For example, copy and paste commands from  [Enable required repositories](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#id4)[](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#enable-required-repositories) will result in following error: 

```
➜  ~ sudo apt update && sudo apt install curl -y
curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
sudo apt install /tmp/ros2-testing-apt-source.deb
zsh: parse error near `)'
```

More details see https://github.com/ros2/kilted_tutorial_party/issues/526

christophebedard, 2025-05-02:
I'm not very familiar with zsh and how it behaves compared to bash or sh, but I noticed that your command includes some backslashes:

```
$ curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
```

Compare it to the actual command in the instructions:

```diff
-curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
+curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0~$(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
```

If I run the one with the backslashes in a bash shell, I get a similar error:

```
$ curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
bash: syntax error near unexpected token `)'
```

Again, I'm not sure if the backslashes were added automatically when copying the command into your shell. Can you try without the backslashes, and also try adding a backslash before the closing parenthesis (after `VERSION_CODENAME`)?

Juliaj, 2025-05-02:
> Can you try without the backslashes, and also try adding a backslash before the closing parenthesis (after `VERSION_CODENAME`)?

This did the trick. It looks like When I pasted these command, zsh automatically escapes the special chars ( ~ and $) with backslashes and that caused the error.  So this is not issue with documentation rather specific to ZSH shell character escaping.

christophebedard, 2025-05-02:
I think it would be nice to have instructions/commands that also work with zsh, but that depends on what/how many commands we'd need to modify.

Reading the answers to this StackExchange question, it looks like this may be because it's recognized as a URL so some characters get escaped automatically, and quoting the URL using double quotes might prevent the automatic escaping: https://unix.stackexchange.com/questions/293472/paste-url-into-terminalurxvt-zsh-failed-some-characters-get-escaped.

Could you try copying this command (with the quoted URL) into your zsh shell and see if it still adds escape characters:

```
curl -o /tmp/ros2-testing-apt-source.deb "https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0~$(. /etc/os-release && echo $VERSION_CODENAME)_all.deb"
```

Juliaj, 2025-05-03:
> Could you try copying this command (with the quoted URL) into your zsh shell and see if it still adds escape characters:

No escaping characters were added. This works!

christophebedard, 2025-05-03:
Thanks for confirming! I've opened #5521 to add the double quotes around the URL

PR #5521 (2025-05-03): Double quote ros-apt-source pkg URL
Fixes #5462

zsh auto-escapes characters when pasting something that looks like a URL, which adds unnecessary escape characters. See https://github.com/ros2/ros2_documentation/issues/5462#issuecomment-2847862418. Double-quoting the URL prevents that.
PR #5522 (2025-05-03): Double quote ros-apt-source pkg URL (backport #5521)
Fixes #5462

zsh auto-escapes characters when pasting something that looks like a URL, which adds unnecessary escape characters. See https://github.com/ros2/ros2_documentation/issues/5462#issuecomment-2847862418. Double-quoting the URL prevents that.<hr>This is an automatic backport of pull request #5521 done by [Mergify](https://mergify.com).
closed 2025-05-03 by christophebedard