ros2/ros2_documentation #5527: 📝 Wrong command in Actions tutorial

SuperGops7, 2025-05-04:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Jazzy

### Description

I was going through the [Creating an Action](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html) tutorial, and noticed the command to source the workspace to have a .console extension for Linux distributions. Shouldn't it be `source install/local_setup.bash`?

### Affected Pages/Sections

https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html#building-an-action

### Screenshots or Examples (if applicable)

[image]

### Suggested Fix

Change the command to `source install/local_setup.bash`

christophebedard, 2025-05-04:
Yeah, that should definitely be `.bash`. Would you be willing to open a PR to fix it?

ahcorde, 2025-05-04:
Related PR is merge https://github.com/ros2/ros2_documentation/pull/5537

closed 2025-05-04 by ahcorde