ros2/ros2_documentation #5547: Document new ros2 action echo command in Rolling / Kilted

kscottz, 2025-05-05:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [x] 🚀 Enhancement

### Distribution

Rolling / Kilted

### Description

I was just going though the Kilted release notes and realized that we don't have documentation for the new `ros2 action echo command.` We should get that out before the release. [The relevant PR is here](https://github.com/ros2/ros2cli/pull/978).

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html#

christophebedard, 2025-05-05:
It's here, but the [Kilted release notes entry](https://docs.ros.org/en/rolling/Releases/Release-Kilted-Kaiju.html#action-introspection) should indeed link to it:

* Rolling: https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html#ros2-action-echo
* Kilted: https://docs.ros.org/en/kilted/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html#ros2-action-echo

kscottz, 2025-05-05:
Cool. Thanks @fujitatomoya. This was a reminder to myself to take care of it today or tomorrow, but if you have a container spun up already, by all means.

closed 2025-05-06 by ahcorde