ros2/ros2_documentation #5550: Add recent ros bag play changes to CLI documentation

kscottz, 2025-05-06:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [x] 🚀 Enhancement

### Distribution

Kilted, Rolling

### Description

I am working my way through the Kilted release notes and I noticed that we added two new features to ROSBag2.

* [Multiple bag playback](https://github.com/ros2/rosbag2/pull/1848)
* [New bag playback status bar](https://github.com/ros2/rosbag2/pull/1836)

We are missing these two features in our [CLI documentation for ROSBag2](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html#). The new `-i` and `--input` commands are particularly useful and should be documented. 

### Affected Pages/Sections

https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html#

kscottz, 2025-05-06:
[There are other new features that need to be documented as well.](https://docs.ros.org/en/rolling/Releases/Release-Kilted-Kaiju.html#action-introspection-rosbag2-support)

chen-harrison, 2025-06-25:
I'd like to try working on this!

christophebedard, 2025-06-26:
Great! Let us know if you have any questions.

chen-harrison, 2025-07-08:
@kscottz @christophebedard while making changes to this page in the documentation, I noticed some other things in ROS 2 bag documentation that were outdated:

- The explanation that ros2 bag record will generate a file, when now by default it creates a recording directory and then adds files of the same name + indices inside of it
- When I use positional arguments for topics in Kilted, it now warns that this is deprecated, so should the documentation teach the use of `--topics`?
- Terminal output for the varying `ros2 bag` commands have changed
- etc.

Would it be better to create a different PR for these changes, or roll them all into one with the new additions describing multiple bag play with `-i`?

kscottz, 2025-07-08:
@chen-harrison thanks so much for helping out with this.

Generally speaking smaller PRs are better. If you have the first task done let's review that and get it out the door before proceeding. Having said that, if this is all in one documentation file that just needs a comprehensive facelift we're probably find doing all the changes at once.

PR #5834 (2025-08-29): Add new `ros2 bag` Kilted features + updates (backport #5771)
## Description

- Demonstrate the new `ros2 bag play` progress bar
- Provide instructions on how to use new `-i` argument to play multiple bags
- Add an example of recording multiple bags earlier in the tutorial, so as to have an example to use with `-i` explanation
- Add `--topics` to `ros2 bag record` commands to reflect the deprecation of positional topic arguments
- Correct wording so that document is consistent with explaining that a new bag **directory**, not file, is made (with the given `-o` argument, if provided), and pathi [... cut]
closed 2025-08-29 by christophebedard