ros2/ros2_documentation #5557: 📝 Clarify Connext RMW instructions also support ARM64

fgallegosalido, 2025-05-06:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Rolling

### Description

The documentation to install RTI Connext states the following:

- ROS 2 users can install a non-commercial-use version of the RTI Connext DDS libraries for x86_64 Linux from the ROS apt repository using the following command:

We should substitute it for the following (as suggested inhttps://github.com/ros2/kilted_tutorial_party/issues/283#issuecomment-2850748081) :

- ROS 2 users can install a non-commercial-use version of the RTI Connext DDS libraries from the ROS apt repository using the following command:

Also, the target platforms doesn't include arm64 as a supported platform for Ubuntu. We should also correct that. See the link: https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Installation/RMW-Implementations/DDS-Implementations/Working-with-RTI-Connext-DDS.html, https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026

### Suggested Fix

Change the line right after the section "ROS 2 apt repositories":

- ROS 2 users can install a non-commercial-use version of the RTI Connext DDS libraries for x86_64 Linux from the ROS apt repository using the following command

for:

- ROS 2 users can install a non-commercial-use version of the RTI Connext DDS libraries from the ROS apt repository using the following command

### Additional Context

This fix is also related to the fix in the target platforms. See https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026

fgallegosalido, 2025-05-06:
@fujitatomoya Where can we update the target platforms? The package `rmw_connextdds` works in arm64

fujitatomoya, 2025-05-06:
@fgallegosalido 

we had a chat about this in today's ROS PMC meeting.
the current status is that it just happens to be working, and we do not think we are ready to support arm64 right away.
for supporting this officially, we need to make sure that does not break other things for sure, and there also needs to be vote for PMC REP 2000.

@mjcarroll @cottsay please chime in if you have anything else to say.

fujitatomoya, 2025-05-07:
> Also, the target platforms doesn't include arm64 as a supported platform for Ubuntu. We should also correct that. See the link: https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026

because this issue to have official support in REP 2000, i will reopen this. (see https://github.com/ros2/ros2_documentation/issues/5557#issuecomment-2855329160)

fgallegosalido, 2025-05-08:
Thanks @fujitatomoya! We will try and work on that during the L release cycle to have ARM 64 fully supported.

fgallegosalido, 2025-05-20:
Hi @fujitatomoya, we realized that in the rep-2000 page, in the RMW Middleware list for Kilted Kaiju, the Connext version is sais it's 6.0.1.25 (https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026), but it should be 7.3.0. Can we amend that?

christophebedard, 2025-05-20:
I opened a PR for that: https://github.com/ros-infrastructure/rep/pull/420