ros2/ros2_documentation #5566: Re implement message field links

EmmanuelMess, 2025-05-06:
Under the old documentation, the types had links, so that one could navigate to a field's type, directly from the message, now it doesn't. 

For example: Going from https://docs.ros.org/en/ros2_packages/jazzy/api/sensor_msgs/msg/Imu.html to https://docs.ros.org/en/ros2_packages/jazzy/api/std_msgs/msg/Header.html is complex.
But going from http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html to http://docs.ros.org/en/noetic/api/std_msgs/html/msg/Header.html is just a single click.

christophebedard, 2025-05-06:
This would definitely be great to have. Note that it belongs in the [ros-infrastructure/rosdoc2 repo](https://github.com/ros-infrastructure/rosdoc2), which is the tool that generates API documentation, including interface pages like: https://docs.ros.org/en/jazzy/p/std_msgs/msg/Header.html.

I believe this was discussed before, but I can't find an issue about it, so it would be worth just opening the same issue on that repo!

EmmanuelMess, 2025-05-07:
I thought I had posted it to rosdoc2 😅 .

Moved to https://github.com/ros-infrastructure/rosdoc2/issues/195.

closed 2025-05-07 by EmmanuelMess