ros2/ros2_documentation #5570: 📝 Zenoh not mentioned in middleware documentation

sea-bass, 2025-05-07:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Kilted + Rolling

### Description

Now that `rmw_zenoh` is officially supported as Tier 1 starting with ROS 2 Kilted, the [Concepts / Intermediate Concepts / Different ROS 2 middleware vendors](https://github.com/ros2/ros2_documentation/blob/rolling/source/Concepts/Intermediate/About-Different-Middleware-Vendors.rst) doc page needs to be updated to not only include Zenoh, but to not make it seem like DDS is the only middleware around.

### Affected Pages/Sections

https://docs.ros.org/en/kilted/Concepts/Intermediate/About-Different-Middleware-Vendors.html

fujitatomoya, 2025-05-07:
agree on this, we can update/add rmw_zenoh information under `Concept` where only mentions about DDS-based RMW implementations. we have updated installation pages like https://docs.ros.org/en/rolling/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html only. i think this is `help-wanted` and `good-first-issue`.

newfadel, 2025-05-08:
Looks like the documentation pages [About-Different-Middleware-Vendors.rst](https://github.com/ros2/ros2_documentation/blob/rolling/source/Concepts/Intermediate/About-Different-Middleware-Vendors.rst) and [About-Middleware-Implementations.rst](https://github.com/ros2/ros2_documentation/blob/rolling/source/Concepts/Advanced/About-Middleware-Implementations.rst) need updates to reflect `rmw_zenoh`'s Tier 1 status.

You could add a new section or bullet point for Zenoh in these `.rst` files. For example, in `About-Different-Middleware-Vendors.rst`, you could add an entry similar to how other vendors are listed, detailing `rmw_zenoh`.

Adding a snippet like this in the relevant section could work:
```rst
* **Zenoh (`rmw_zenoh`)**: Provide a brief description and link to the `rmw_zenoh` documentation.
```
This would involve editing the reStructuredText source files directly. Let me know if you have questions about the `.rst` syntax!

sea-bass, 2025-05-09:
@newfadel tell ChatGPT thanks on my behalf!

aehils, 2025-05-10:
I'd be happy to help! I'll get these updated :)

fujitatomoya, 2025-05-10:
@aehils thanks, i will go ahead to assign this to you.

PR #5666 (2025-05-24): Update middleware documentation to include Zenoh (backport #5644)
Fixes #5570 

## Description

With the promotion of `rmw_zenoh` to Tier 1 support in ROS 2 Kilted, it became necessary to reflect this in relevant doc pages. The aim is also to provide information about the Zenoh middleware and to avoid implying DDS is the only protocol option for middleware implementations. 

The changes target these pages:

https://docs.ros.org/en/kilted/Concepts/Intermediate/About-Different-Middleware-Vendors.html
https://docs.ros.org/en/kilted/Concepts/Advanced/About-Middleware-Implementations.html

### Did  [... cut]