ros2/ros2_documentation #5571: 📝 Discuss Events executor in "About Executors" page

sea-bass, 2025-05-07:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Kilted / Rolling

### Description

The [Concepts / Intermediate Concepts / Executors](https://github.com/ros2/ros2_documentation/blob/rolling/source/Concepts/Intermediate/About-Executors.rst) doc page does not currently mention the `EventsExecutor`, which has been around for a bit now and also has an experimental Python version available as of Kilted.

I don't know enough about executors to do it justice, but I wanted to leave this issue here for someone who may.

### Affected Pages/Sections

https://docs.ros.org/en/kilted/Concepts/Intermediate/About-Executors.html

sea-bass, 2025-05-11:
This sounds great @fujitatomoya -- I have not yet used `EventsExecutor` myself, but maybe it's still worth putting it in the main documentation with a parenthetic "(experimental!)" on the side?

This way people can try it (and find bugs!), and if/when it is fully promoted, the doc PR would be to simply remove a disclaimer?

fujitatomoya, 2025-05-11:
i cannot disagree with that!

mvanhorn, 2026-10-30:
The executor overview listed only the standard executor types and had no section describing the experimental EventsExecutor implementations or their status.

The executor hierarchy now includes EventsExecutor alongside the standard types. The page explains its event-driven design, C++ namespace and usage, experimental API status, deprecation in favor of EventsCBGExecutor in Rolling, and distinction from the Callback Group Events Executor.

https://github.com/ros2/ros2_documentation/pull/7381