ros2/ros2_documentation #5597: 📝 GitHub added a new type of access token (fine-grain) bloom-release doc page not covering, results in confusion

130s, 2025-05-09:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

(I think this issue is ros-distro agnostic)

### Description

Github's Fine-grained PATs has been [introduced sometime in 2022 or 2023](https://github.blog/changelog/2025-03-18-fine-grained-pats-are-now-generally-available/). On each account's PAT generation page https://github.com/settings/tokens there are 2 options, classic or newly added fine-grained tokens. But the ros2 doc not covering confuses readers (like it did to me).

From my experience, fine-grained token doesn't seem to work (yet) with `bloom-release`, so only option is the `classic`. If that's true, document should clarify that.

130s, 2025-05-09:
With fine-grained PAT, bloom still failed with code `403`.

```
$ bloom-release --pull-request-only --rosdistro $ROSDISTRO --track $ROSDISTRO $REPO_NAME
==> Fetching 'rqt_moveit' repository from 'https://github.com/ros-gbp/rqt_moveit-release.git'
Cloning into '/tmp/tmpt7ek_gre'...
remote: Enumerating objects: 1487, done.
remote: Counting objects: 100% (576/576), done.
remote: Compressing objects: 100% (427/427), done.
remote: Total 1487 (delta 190), reused 509 (delta 136), pack-reused 911 (from 1)
Receiving objects: 100% (1487/1487), 183.47 KiB | 4.27 MiB/s, done.
Resolving deltas: 100% (475/475), done.
==> Generating pull request to distro file located at 'https://raw.githubusercontent.com/ros/rosdistro/master/noetic/distribution.yaml'
Unified diff for the ROS distro file located at '/tmp/tmp3l05vf42/rqt_moveit-0.5.13-1.patch':
--- noetic/distribution.yaml
+++ noetic/distribution.yaml
@@ -10713,7 +10713,7 @@
       tags:
         release: release/noetic/{package}/{version}
       url: https://github.com/ros-gbp/rqt_moveit-release.git
-      version: 0.5.11-1
+      version: 0.5.13-1
     source:
       type: git
       url: https://github.com/ros-visualization/rqt_moveit.git
==> Checking on GitHub for a fork to make the pull request from...
Could not find a fork of ros/rosdistro on the 130s GitHub account.
Would you like to create one now?
Continue [Y/n]? 
Aborting pull request: HTTP Error 403: Forbidden (https://api.github.com/repos/ros/rosdistro/forks): None
The release of your packages was successful, but the pull request failed.
Please manually open a pull request by editing the file here: 'https://raw.githubusercontent.com/ros/rosdistro/master/noetic/distribution.yaml'
❌  No pull request opened.
```

After I swapped the PAT locally stored in `~/.config/bloom`, the same cmd above went through (resulted in opening a PR https://github.com/ros/rosdistro/pull/45677).

130s, 2025-05-09:
I don't know, but [stackoverflow.com#67628262](https://stackoverflow.com/questions/67628262/oauth-scope-required-for-creating-github-pull-requests-with-personal-access-toke) indicates that with fine-grained PAT one needs write-access on a repo in order to open a PR (which is not always possible e.g. a pkg maintainer shouldn't need a write access on `ros/rosdistro` in order to enlist her pkg on rosdistro).

christophebedard, 2025-05-22:
For what it's worth, we ran into similar limitations with the new tokens with https://github.com/ros-tooling/ros-github-scripts and also just changed the doc to require a classic token: https://github.com/ros-tooling/ros-github-scripts/pull/24

130s, 2025-05-22:
> The guide says: https://docs.ros.org/en/rolling/How-To-Guides/Releasing/First-Time-Release.html#set-up-a-personal-access-token
> 
> > 3. In the dropdown, select **Generate new token (classic)**

Ok, good to know it's clarified in `rolling` doc.

I guess my problem may be specific with Google search. The 1st search result as of today for ["ros2 release token"](https://www.google.com/search?q=ros2+release+token&rlz=1CAVXIY_enUS1081&oq=ros2+release+token&gs_lcrp=EgZjaHJvbWUyBggAEEUYOTIGCAEQRRg8MgYIAhBFGDzSAQk0NTI4NGowajGoAgCwAgA&sourceid=chrome&ie=UTF-8&sei=MHUvaKjsGcfIp84PzbSEyA0) (note I'm using Chrome's Incognito mode in an attempt to eliminate personalization) returns [`foxy` version](https://docs.ros.org/en/foxy/How-To-Guides/Releasing/First-Time-Release.html), which does NOT yet include the clarification (and that's what I referenced to, hence confusion and opening this ticket).

[image]

christophebedard, 2025-05-31:
We unfortunately can't really do much about that. There is a warning at the top of every EOL distro page:

[image]

Otherwise, we hope that eventually Google switches to newer versions of the docs as they grow and have more links to/from them.

Given the above, I'll close this PR. Feel free to keep discussing.

closed 2025-05-31 by christophebedard