ros2/ros2_documentation #5601: 📝 Fix issues with building CycloneDDS from scratch

jncfa, 2025-05-09:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Distribution

Kilted

### Description

The instructions for compiling rmw_cyclonedds use the cyclonedds main branch, which is not guaranteed to be working at all times, as was the case during the kilted testing party.

### Affected Pages/Sections

https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html

### Suggested Fix

I would suggest fixing the branch of the cyclonedds repository, which currently seems to be 0.10.5.

### Additional Context

Relevant issue: https://github.com/ros2/kilted_tutorial_party/issues/545#issuecomment-2866458832

christophebedard, 2025-05-14:
Thanks for creating this issue. I think we should just tell users to look at the `version:` under `eclipse-cyclonedds/cyclonedds:` in the distro's `ros2.repos` file and check out that version, e.g., `releases/0.10.x` for Rolling: https://github.com/ros2/ros2/blob/rolling/ros2.repos.

Would you be willing to open a PR with this change?

jncfa, 2025-05-15:
> Thanks for creating this issue. I think we should just tell users to look at the `version:` under `eclipse-cyclonedds/cyclonedds:` in the distro's `ros2.repos` file and check out that version, e.g., [...]

Sure!

PR #5678 (2025-05-27): Indicate correct branches to checkout to build CycloneDDS from source
## Description

Adds a minor description on how to determine the correct versions to checkout to build CycloneDDS from source.

I also added a minor script which will automatically fetch the correct branch version for Cyclone DDS.

Fixes #5601

### Did you use Generative AI?

No

### Additional Information
closed 2025-05-27 by christophebedard