ros2/ros2_documentation #5667: 📝 Coloring in ROS2 distribution tables is wrong (?)

jlblancoc, 2025-05-24:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Description

I think the green rows should be 1st, 2nd and 4th, right? If you confirm I could prepare a PR to fix it: 

### Screenshots or Examples (if applicable)

[image]

### Suggested Fix

Change coloring

christophebedard, 2025-05-24:
Thanks for reporting! #5668 will fix this.

christophebedard, 2025-05-24:
> I think the green rows should be 1st, 2nd and 4th, right? If you confirm I could prepare a PR to fix it:

Ah, sorry, I just did it! It should actually be 2nd and 4th; Kilted is not an LTS release.

PR #5668 (2025-05-24): Fix green rows in distributions table
## Description

Fixes #5667

The indexes of the rows to be marked in green needed to be updated.

### Did you use Generative AI?

### Additional Information
PR #5670 (2025-05-24): Fix green rows in distributions table (backport #5668)
## Description

Fixes #5667

The indexes of the rows to be marked in green needed to be updated.

### Did you use Generative AI?

### Additional Information

<hr>This is an automatic backport of pull request #5668 done by [Mergify](https://mergify.com).
closed 2025-05-24 by christophebedard