ros2/ros2_documentation #5761: Extend how to manipulate parameters at runtime to mention ability to do so from client libraries

tfoote, 2025-06-30:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [x] 🚀 Enhancement

### Generated by Generative AI

No

### Distribution

Rolling et al

### Description

The parameter capabilities include the ability to interact with them at runtime from all the client libraries. 

However the default parameter documentation doesn't mention that in the section: [Manipulating parameter values at runtime](https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#id9)[](https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#manipulating-parameter-values-at-runtime)

https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#manipulating-parameter-values-at-runtime

https://github.com/ros2/ros2_documentation/blob/c8a8a89eb78503f22381bb9ee03c0b748d414255/source/Concepts/Basic/About-Parameters.rst?plain=1#L113C3-L119

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#manipulating-parameter-values-at-runtime

### Suggested Fix

It would be good to add documentation to mention that and give pointers to some of the client libraries. 

### Additional Context

This came from this question where the asker didn't know that there was convenient ways to set parameters: https://robotics.stackexchange.com/questions/113686/c-equivalent-of-ros2-params-load

akoushik2k, 2025-07-13:
Hi, I’d love to take this on as my first open-source contribution!
I’m new to contributing but eager to learn and get involved.

I noticed that the [Using ros2 param](https://docs.ros.org/en/rolling/How-To-Guides/Using-ros2-param.html) page is already linked in the documentation section referenced in this issue. Before I proceed, I wanted to ask:

Would you like this section to be expanded to give more context about client library support (e.g., rclcpp, rclpy) and runtime parameter APIs?

Or is the goal mainly to ensure this functionality is clearer and more discoverable within the "Manipulating parameter values at runtime" section?

I’d appreciate your guidance so I can align the update with the intent of the issue. Looking forward to contributing!

Humobot1812, 2025-07-25:
@tfoote Please assign this issue to me I have the repo setup on my system

closed 2026-01-09 by fujitatomoya