ros2/ros2_documentation #5772: Update C++ style guide with enum / enum class naming

oysstu, 2025-07-09:
### Issue Type

- [x] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Description

This might need a broader discussion on naming. Currently, there is a mix of snake_case, PascalCase, and UPPER_CASE being used. All of these deviate from the referenced google style guide, which prefers kPascalCase.

Should a notice be added about this deviation in the documentation?

anishk85, 2025-10-05:
hi i added this in the following pr #5918 
kindly merger this pr

anishk85, 2025-10-05:
@mjcarroll @asymingt