ros2/ros2_documentation #5773: 📝"Testing with pre-release binaries" for Ubuntu Noble outdated?

knmcguire, 2025-07-11:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [x] 📖 Outdated Content
- [ ] 🚀 Enhancement

### Generated by Generative AI

N/A

### Distribution

Jazzy, Kilted, Rolling

### Description

[The page of testing](https://docs.ros.org/en/rolling/Installation/Testing.html) with pre-release binaries, still mentions that we have to change the ros2.list (`/etc/apt/sources.list.d/ros2.list`). In Ubuntu 24.04 they changed this to a new format, so now it's in `ros2.sources` and in a different format.

I couldn't really find a source that mention this change, but perhaps good to change the documentation for just rolling/kilted/jazzy since those are focused on Noble?

### Affected Pages/Sections

https://docs.ros.org/en/rolling/Installation/Testing.html

### Screenshots or Examples (if applicable)

It's just step 2 in deb testing repository

Edit (with sudo) the file /etc/apt/sources.list.d/ros2.list and change ros2 with ros2-testing. For example, on Ubuntu Noble the contents should look like the following:

# deb http://packages.ros.org/ros2/ubuntu noble main
deb http://packages.ros.org/ros2-testing/ubuntu noble main

### Suggested Fix

Edit (with sudo) the file /etc/apt/sources.list.d/ros2.source and change ros2 with ros2-testing. For example, on Ubuntu Noble the contents should look like the following:

# URIs: http://packages.ros.org/ros2/ubuntu
URIs: http://packages.ros.org/ros2-testing/ubuntu

christophebedard, 2025-07-11:
Maybe @claraberendsen can confirm, but that seems correct! I assume it changed as part of the new ros-apt-source package/setup: https://github.com/ros-infrastructure/ros-apt-source/blob/main/ros-apt-source/debian/ros2-apt-source.links.

claraberendsen, 2025-07-17:
PR opened, it seems I missed porting changes over from the Kilted docs, that one has the corresponding updates.

PR #5783 (2025-07-18): Update testing installation docs to use ros-apt-source package

## Description

This PR unifies the deb testing repository instruction to those on the [Kilted docs](https://docs.ros.org/en/kilted/Installation/Testing.html).

Additionally a NEW section with the RPM instructions is added to indicate how to do the switch between testing and main on rpm. 

Fixes #5773 

### Did you use Generative AI?

I did not use any generative AI.

### Additional Information
These changes should be ported over to jazzy, humble and kilted.
PR #5789 (2025-07-18): Update testing installation docs to use ros-apt-source package (backport #5783)

## Description

This PR unifies the deb testing repository instruction to those on the [Kilted docs](https://docs.ros.org/en/kilted/Installation/Testing.html).

Additionally a NEW section with the RPM instructions is added to indicate how to do the switch between testing and main on rpm. 

Fixes #5773 

### Did you use Generative AI?

I did not use any generative AI.

### Additional Information
These changes should be ported over to jazzy, humble and kilted.

<hr>This is an automatic backport of pull request #5783 [... cut]
PR #5788 (2025-07-24): Update testing installation docs to use ros-apt-source package (backport #5783)

## Description

This PR unifies the deb testing repository instruction to those on the [Kilted docs](https://docs.ros.org/en/kilted/Installation/Testing.html).

Additionally a NEW section with the RPM instructions is added to indicate how to do the switch between testing and main on rpm. 

Fixes #5773 

### Did you use Generative AI?

I did not use any generative AI.

### Additional Information
These changes should be ported over to jazzy, humble and kilted.

<hr>This is an automatic backport of pull request #5783 [... cut]
closed 2025-07-18 by fujitatomoya