ros2/ros2_documentation #5839: Use  literalinclude:: For Code Examples / Tutorials

InvincibleRMC, 2025-09-02:
A lot of the code examples like `minimal_publisher` are just copied from ros2_examples repo. If we could read the raw files from either the built deb, a submodule or, just downloading off of GitHub. We then could use [literalinclude](https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.rst) to avoid the duplication. The end goal would be to reduce maintainer burden of having to manual copy code.

fujitatomoya, 2025-09-03:
There has been discussion on this enhancement.
I was hoping that [literalinclude](https://www.sphinx-doc.org/en/master/usage/restructuredtext/directives.html#directive-literalinclude) could support the URL, so that we can just point the URL for the source code with appropriate branch, but i do not think that it supports that.
for now, i do not think of any other options instead of downloading them from GitHub as you suggested.
I believe it is still great enhancement to reduce maintenance burden.

InvincibleRMC, 2025-09-03:
The job is generate from [here](https://github.com/ros-infrastructure/ros_buildfarm/blob/ff7c1cf38362d9fecae8ef2b677719e2d8ddc04b/ros_buildfarm/templates/doc/doc_independent_job.xml.em). I'm not familiar with the build farm and its functions. But I imagine we could add a clone step to get the raw files then use literalinclude with some function in `conf.py` to determine file location.

gavanderhoorn, 2025-09-09:
> thanks for pointing that PR and discussion. bad memory... sorry about that 😓 

no need to apologise. I only remembered because it was my own PR (and issue).

> that makes sense to track the related source code change which is referred by the documentation, otherwise it can be broken documentation. besides that, backports could be more difficult to resolve the conflict.

as I wrote earlier: it's a trade-off by documentation becoming stale due to untracked/unnoticed changes and 'freezing' documentation to make sure it documents a particular 'state'.

> those requirements also come to #5840... maybe we should give another look at #2828?

perhaps, I believe a choice should be made about what we prefer, and then we can decide how to implement it.

PS: re: stale due to changes not being picked up on: I had (a draft of) a script that compared the hashes passed to `remote-literal-include` with the `HEAD` of the specific branch they belonged to. If they didn't match, it would output a `WARN`. Including that in a (scheduled) CI run could help improve visibility of documentation 'breaking'.